rLoopTeam/eng-software-pod #20: Design and test the I2C read and WRITE interface

omegabytes, 2016-10-25:


edward6chan, 2016-10-28:
I am a little confused. This issue is to be able to read from pressure sensor and write to PI??? While #6  is to be able to read from temp sensor. What is coming via the RX? How are we converting? 

My understanding is that each RM48 will have three DCAN interfaces that provide a high-speed serial interface. So, say, we would be reading temp readings via J2 and then we would be reading pressure sensor via J3?

rjvyas, 2016-10-28:
@edward6chan good question. We need clarification if whether we are writing to Pi or GS for this issue. And you are right, this issue is relating to the pressure sensor but if the read write interface is the same as #6 then the only thing different is reading/writing pressure sensor as opposed to temp sensor.

edward6chan, 2016-10-31:
This issue is related to #168 , not #6

jennylin, 2016-11-08:
Acceptance Criteria:
Acceptance Criteria;

- With MS5607 connected to RM48, show that the ground station is receiving pressure from MS5607 through the RM48 and Pi
- Apply pressure on MS5607 and watch number go up
- Release pressure on MS5607 and watch number go down
- Disconnect everything and retrieve and verify dumped data

SafetyLok, 2016-11-30:
Done.

closed 2016-11-30 by SafetyLok