rLoopTeam/eng-software-pod #24: Provide a proper reset function

omegabytes, 2016-10-25:


edward6chan, 2016-10-30:
@SafetyLok 
I am trying to read and write bytes to MS5607. 
It appears that _rm4_i2c__user.c_ is designed for SMBus device; not compatible with MS5607.

I'd need to use _rm4_i2c.h_ functions directly (I imagine) - would that be okay? My understanding is that that's IP...

And if that's okay, I would need to define additional state machines to avoid wait milliseconds. I'll be using

- Set Start
- Send Bytes w/ the device address
- Send Bytes again w/ the command code
- Send Stop (I don't know the differences between Clear_Stop vs Set_Stop)
(Not understanding clear stop)

Would DMA actually be easier...? But I suppose that's reserved for more "important" task when we need optimization?

edward6chan, 2016-10-31:
- [x] Completed
`void vMS5607__Reset(void)
{
    vMS5607__Write8(MS5607_CMD__RESET);
}`

edward6chan, 2016-11-06:
Using **Lint16 s16RM4_I2C_USER__TxReg	(device address,register address)** from _rm4_i2c__user.c_. This issue should be ready to test

SafetyLok, 2017-08-16:
Done, device been working.

closed 2017-08-16 by SafetyLok