rLoopTeam/eng-software-pod #6: Design and test I2C read interface

omegabytes, 2016-10-25:
all this needs is a definition!

edward6chan, 2016-10-28:
Can someone look at power-node-comm branch and provide feedback?

edward6chan, 2016-10-28:
Some codes @rjvyas wrote for onReceived Event

`+void onReceivedDecParam(struct rI2CRX_decParam decParam) //Pi sending data to power node in struct decParam. Receiving data from Pi and coding RM48
 {
-    printf("TODO");
+    if (decParam.index == 5) //Check if received value for index is 5. 5 is an arbitrary number
+    {
+        uint8_t* t_ptr = (uint8_t*)decParam.val; // force compiler to think the data located at val is a 8 bit integer
+        uint8_t t = *t_ptr; // retrieve the data from that location
+        printf("shutdown %d\r\n", t);
+        if (t == 0) {
+            printf("shutdown battery power\r\n");
+        }
+        else if (t == 1) {
+            printf("shutdown power node itself\r\n"); // t= 0 and 1 decision is completely arbitrary
+        }
+
+        // resets the packet building state machine
+        rI2CTX_beginFrame();
+        // start building the packet
+        rI2CTX_addParamter_uint8(5, t); // this is just a echo as a form of acknowledgement
+        rI2CTX_endFrame(); // this actually sends it
+    }
+        
 }`

omegabytes, 2016-10-28:
SHA: 07916f01ac79862578a00a34747d9c0f4041ed04

edward6chan, 2016-10-28:
This should also be renamed to UART at some point, as we are communicating ffrom RM4 to Pi via UART serial. And, we need to handle async, right?

piense, 2016-10-28:
@edward6chan Yeah, I2C is old. What do you mean by "handle async"?

I added some comments on her branch. I'll put them here too:
Mostly good. The echo part would wreck havoc though, and rI2CTX_endFrame() doesn't actually send a packet. It just finalizes the packet footer in RAM. There will only be one function in the entire node that creates and sends packets. If the GS needs validation of a command then the results of that command should be reflected somewhere in the feedback data we get from the pod.
Not sure about the printfs either. On the RM48 they wouldn't go anywhere. Maybe wrap it in some WIN32 conditions for debugging in the emulator?

rjvyas, 2016-10-28:
@piense The printfs are for debugging in the emulator. I still need to test it in the emulator. And need to clean up the code to adhere to Lachlan's code standard.

rjvyas, 2016-10-28:
@edward6chan Ed, async for UART means that since there is no clock, we are constantly streaming the data. There is no synchronization, like in I2C, where there is master-slave communication such that the data is only sent when ready. With UART, if we are not constantly streaming, we could lose data

edward6chan, 2016-10-28:
@rjvyas understood. I was just trying to figure out how this would work in terms of baud rate and framing the bit. Perhaps I was overthinking. 

This function only has to read? Do we have to specify serial port somewhere? The conversion to actual temperature will happen later?

Well, I guess we are still stuck at "design" It's still a little unclear to me what needs to be written

piense, 2016-10-28:
@rjvyas There isn't a packet queue at all. It'd be fairly resource intensive to do on an embedded system like this, especially without hardware support like there is for the CAN bus. The idea is to basically just dumps all relevant data at a certain fps. For something like that power node with 540 temperature sensors we can go at say 30-100 fps and stagger which temp sensors are in which frame and the important bits can go every time. It is a binary protocol though it isn't all that efficient for bandwidth. It was designed to be framed pretty nicely for easy processing on the RX side without having to keep the code on both sides perfectly in sync as far as the list of data parameters coming across. There's no mark or break conditions to deal with on the wire. So you can factor in 50% overhead on top of the raw number of bytes of the data, bigger data types have less overhead. So 50x 16-bit parameters would be about 200 bytes on the wire. So 1600 bits per frame, which puts you at a max refresh rate of 625 fps at 1Mbps. Of course it's hard to the frames exactly back to back on the wire without a real hardware packet buffer, but with DMA we can actually get fairly close and in this case bigger packets are actually more efficient. Technically you also need a bit of extra bandwidth for escaping special characters in the stream since the control characters are embedded in the byte stream. But worst case a frame is a few microseconds late, shouldn't be any data loss if the TX function is coded correctly and just double checks the previous DMA is clear before going again.

@edward6chan As for async or not is essentially beyond the scope of the PiComms module. The comms code takes care of formatting the packet for the wire. It was originally written for I2C bus but got moved to the UART when we jumped to the RM48s. We could go clean up those names. But the lower layers don't matter to the comm code, it just hands off a byte buffer and the lower drivers are responsible for getting it out the hardware.

edward6chan, 2016-10-30:
@piense I would like to be able to "connect up some Pi code to the DLL and emulate sending a packet from DLL -> PC serial port -> Pi. Make sure this system works, log results against code version."

What would you still need to complete so I can proceed?

On my side, I think I have @rjvyas 's code that is creating the packet. I will still have to figure out how to send it. Do I need to write this? Or, is there a function I can call? 

@SafetyLok Am I on the right track? This whole thing is still a little ambiguous to me.

> @40chans Those instructions mostly work. The work isn't merged back to the dev branch yet. You can check it out in https://github.com/rLoopTeam/eng-software-pod/tree/piense-gs-comms-wip The only gotcha should be that in the ground station project you need to disable the test data from starting up. It doesn't support data coming from multiple sources on the same PC yet. The ports conflict and things crash.

> Instructions:
1. Compile LDLL175
2. Start LAPP187
3. Click the test button.
4. At the bottom of the screen will be the string of data (hex values) that are transmitted out of the serial port.
5. We need to check that the packet format is correct. We should be able to copy those hex values and paste directly into GS and something should happen.
6. From here we need to take those hex bytes and spit them out a ZeroMQ socket, across the network to the GS.

piense, 2016-10-30:
If you really want to go out the serial port you'll need a USB to TTL adapter of some sort. Then you can grab the VB TX callback in the form as redirect that output to a serial port. There should be a ton of examples online of using the serial port in VB. From there you need a Pi2 to connect the USB to TTL adapter to. You'll also need the firmware for the SD card. I'm out of town at the moment but I may have an image of it available without having to entirely rebuild from scratch. I just won't be able to test it.

rjvyas, 2016-10-30:
@piense Just letting you know, Ed and I working on this together so we will alternate in asking questions.Next question: 
Could we please get the Pi image from you and instructions on how to use it on the Pi?

m4rol, 2016-10-30:
updated the name of the task to reflect the switch from I2C to UART. pls update the variable names

piense, 2016-10-30:
@rjvyas I'll get it to you some time tomorrow afternoon. I'm out of town and don't have it handy. It's not that hard to rebuild from scratch if you have a Ubuntu machine handy but probably not worth fussing with at the moment.

edward6chan, 2016-10-31:
This doesn't belong to TSY01 milestone. This should be under PI_COMMs if a new milestone needs to be created.

This issue will need to be created for TSY01
> The TSYS01 device needs a layer written to comm with the I2C channel to be able to issue an address and read a register.

m4rol, 2016-10-31:
completely agree with @edward6chan @alexgaoo our GH master can u make a new milestone PICOMMS and put this issue under it ?

SafetyLok, 2016-11-30:
Done.

closed 2016-11-30 by SafetyLok