rLoopTeam/eng-software-pod #39: Ensure each of the two devices on the power node PCB works correctly, datalog the results

omegabytes, 2016-10-26:
this is too broad

m4rol, 2016-10-30:
Description of DS2482 https://www.maximintegrated.com/en/products/interface/controllers-expanders/DS2482-100.html

SafetyLok, 2016-10-31:
The power node has two DS2482 devices on board. They share the same I2C bus, but have different addresses. The purpose is that we will split some of the 1-wire devices onto one bus, and some other DS18B20 devices onto other bus.  It will help a bit with redundancy.

We need to verify that both devices work correctly together in hardware. We need to datalog (using an iLogic or otherwise) to ensure that the I2C bus time is shared correctly.

From the software viewpoint we need to ensure that the devices which have been enumerated and are tagged to one 1-wire bus are not confused with the other bus, as a search failure will take time.

rjvyas, 2016-11-06:
@SafetyLok could you please check @nemimccarter's code in #40. I think this issue is already address in the code: if (s16DS2482S_SEARCH__Search(u8DeviceIndex))

SafetyLok, 2016-11-06:
Need to test this on actual hardware with all the sensors loaded.

closed 2017-01-31 by millipedeHz