rLoopTeam/eng-software-pod #43: Create a function to use the 1-wire search system to identify a chain of 1-wire devices.

omegabytes, 2016-10-26:
Log this data to the terminal.

omegabytes, 2016-10-30:
[1-wire search algorithm](https://www.maximintegrated.com/en/app-notes/index.mvp/id/187)

rjvyas, 2016-10-31:
hasn't this algorithm already been implemented in: https://github.com/rLoopTeam/eng-software-pod/blob/4d73dee1bc152cf2613a29a874ecb7222a712170/FIRMWARE/COMMON_CODE/MULTICORE/LCCM641__MULTICORE__DS2482S/SEARCH/ds2482s__search.c

nemimccarter, 2016-11-01:
@rjvyas you are correct, but our implementation `Lint16 s16DS2482S_SEARCH__Search(Luint8 u8DeviceIndex)` appears to be incomplete (some empty `else` statements) and is noted as violating MIRSA 2004 (regulations on using C in critical systems) for its use of `~` and `<<`. I'll check it out!

omegabytes, 2016-11-02:
The following stub exists in ds18b20.c:
```c
//precondition that the DS2482 is operational

	u8ChannelIndex = 0U;
	u8DSIndex = 0U;

	s16Return = s16DS2482S_SEARCH__SearchFirstDevice(u8ChannelIndex, &sDS18B20.sDevice[u8DSIndex].u8SerialNumber[0]);
	if(s16Return >= 0)
	{
		sDS18B20.sDevice[u8DSIndex].u8ChannelIndex = 0U;

		s16Return = s16DS18B20_TEMP__Get_Resolution(u8DSIndex, &sDS18B20.sDevice[u8DSIndex].u8Resolution);

		s16Return = s16DS18B20_TEMP__Request(u8DSIndex, 1U);

		//success
		s16Return = s16DS18B20_SCRATCH__Read(u8DSIndex, &u8Temp[0]);
		if(s16Return >= 0)
		{

		}
		else
		{
			//fall on
		}

	}
	else
	{

	}
```

Am I implementing something similar to this (from ds2482s__search.c: 

```c
if(s16Return > 0)
		{
			//order the address
			pu8Addx[0] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[0];
			pu8Addx[1] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[1];
			pu8Addx[2] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[2];
			pu8Addx[3] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[3];
			pu8Addx[4] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[4];
			pu8Addx[5] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[5];
			pu8Addx[6] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[6];
			pu8Addx[7] = sDS2482S.sSearch[u8DeviceIndex].u8ROM_SerialNumber[7];

		}```

SafetyLok, 2016-11-02:
Lets delay this task. The purpose of it was to search all the busses on our power nodes and the temp sensors.  Currently the DS18B20 and DS2482 already support searching for 1-wire devices, but not an entire pod level search of all systems.

rjvyas, 2016-11-03:
@SafetyLok does this mean that we would also have to delay my task for issue #39 since it also related to bus search?

SafetyLok, 2016-11-30:
Done.

closed 2016-11-30 by SafetyLok