rLoopTeam/eng-software-pod #52: In the sm, read in each device, do this on-by-one per SM process loop, do not try and read all at once (it could take a long time)

omegabytes, 2016-10-26:
In the sm, read in each device, do this on-by-one per SM process loop, do not try and read all at once (it could take a long time)

This sounds like a function needs to be created to do this. Let's redefine it as a task.

jennylin, 2016-11-08:
Acceptance criteria:

- Mark some temperature sensors as ‘bus bar’ and some as ‘battery’ and make sure the configuration will read them as such
- Set threshold for temperature “overload” to something that we can trigger by putting our hands on the sensors for ‘battery’, but a higher temperature that we cannot trigger from hands for ‘bus bar’
- Use oscilloscope or logic analyzer to make sure watchdog is receiving square signals while hands are not on temperature sensors
- Cover a temperature sensor on ‘bus bar’ and make sure we are still receiving square signals
- Cover a temperature sensor on ‘battery’ and make sure we flatline 0
- Reset thresholds so they will both overload
- Make sure both sensors flatline 0
- Reset thresholds so neither overload
- Make sure neither sensors flatline 0
- Repeat in different conditions

SafetyLok, 2016-11-30:
Done.

SafetyLok, 2016-11-30:
Software complete, needs testing and qualification.

SafetyLok, 2017-08-16:
Done, was the source of the slow pod safe command.

closed 2017-08-16 by SafetyLok