rLoopTeam/eng-software-pod #59: Verify the functionality of Undervolt trip configuration

omegabytes, 2016-10-26:
Test if this function works with the IRQ. Log the results.

jennylin, 2016-11-08:
Acceptance Criteria: 

- Using Battery testbed A, from RM48 ask module to discharge and verify voltage has dropped by multimeter
- Using Battery testbed A, from RM48 ask module to discharge and keep discharging until we hit our Undervolt trip threshold (set it a little higher than actual threshold for testing). Verify with multimeter that we stop discharging when threshold is hit
- Using Battery testbed A, start with a battery below Undervolt threshold. Plug in, attempt to discharge, and verify with multimeter that we don’t actually discharge

SafetyLok, 2017-08-16:
ATA device abandoned.

closed 2017-08-16 by SafetyLok