rLoopTeam/eng-software-pod #84: Create a state machine using same techniques in vPWRNODE__Process

omegabytes, 2016-10-26:
A brief outline of the "techniques used in vPWRNODE__Process" goes here. If possible, a code snippet as an example would help devs implement this feature. Desired output/behavior should also be included.

jennylin, 2016-11-08:
Acceptance Criteria:

Battery testbed B: 3 x ATA6870N daisy chained together, each connected to 2 RC batteries (so 6 RC batteries total)

- Take battery testbed B, but with some of the RC batteries only partially charged instead of fully charged (but still above Undervolt trip threshold) and connect to system.
- Use multimeter to make sure state machine is automatically identifying fully charged batteries and discharge until they match the least charged battery.
- From there, make sure they discharge evenly until the Undervolt threshold (or should they stop discharging?) 
- Repeat with different numbers of partially charged batteries and amount of partial charge.

SafetyLok, 2017-08-16:
Done.

closed 2017-08-16 by SafetyLok