rLoopTeam/eng-software-pod #109: Ensure safety limits work in software

omegabytes, 2016-10-28:
(i.e. move planner terminates on interrupt)

SafetyLok, 2016-11-10:
If we move a distance and that distance exceeds the safe travel and we hit the end stop switches we much make sure that the stepper system aborts its current move immediately.

We should check this with a DSO or similar to check the timing in software from interrupt to movement abort. Technically it should be no greater than the minimum step time which is currently 50us.

rjvyas, 2016-11-12:
@SafetyLok I've written functions to issue fault flags if either of the limit switches for the brakes are switched on. Where do I write code that will actually abort the current move immediately based on the fault flags? Which file will that function go in?

SafetyLok, 2017-01-10:
Done, movement planner stops on ISR

closed 2017-01-10 by SafetyLok