rLoopTeam/eng-software-pod #112: Provide a function to change brake position

omegabytes, 2016-10-28:
Provide a function to change brake position based on either Percentage, mm from IBeam, etc.

rjvyas, 2016-11-09:
@SafetyLok Wrote a function to move change the brake position by a percentage. Need some guidance to write the same for changing the position by distance, i.e. between 0 and 25mm. The comment in code says that some calibration is required. Please check the code below for percentage change:

//move the brakes position by a certain percentage between 0-100%
//approx distances are 25mm (fully open) to 0mm (fully closed)
//some calibration will be needed here.
void vFCU_BRAKES__Move_Percent_Position(Lfloat32 f32Percent, E_FCU__BRAKE_INDEX_T eBrake)
{
	if((f32Percent < 0) || (f32Percent > 100))
	{
		//do nothing
	}
	else
	{
		if(eBrake < FCU_BRAKE__MAX_BRAKES)
		{
			//set BrakePosition_Percent to value
			sFCU.sBrakes[(Luint32)eBrake].sMLP.f32BrakePosition_Percent = f32Percent;
		}
	}
}

rjvyas, 2016-11-10:
Need to calculate brake deployment:
 * When the brake is fully deployed the airgap is 2.5 mm.
 * The airgap increases by tan 17 deg x mm moved by lead screw.
 * If counting revolutions, every rotation of the ball screw moves the wedge 3 mm.
 * 0.9172 mm airgap increase per revolution.

SafetyLok, 2017-01-10:
Done, pos can be changed via IBeam or via manual position in development mode.

closed 2017-01-10 by SafetyLok