rLoopTeam/eng-software-pod #121: Interface with N2HET

omegabytes, 2016-10-28:


SafetyLok, 2016-11-28:
The N2HET module in the RM48 allows for uploading a micro instruction set into RAM to execute code in a very precisely timed manor.  

The contrast sensors should make use of the N2HET (for edge detection) as well as the HTU (Highend timer/transfer unit) to capture precise timing for the contrast sensors.  If precise timing is possible we can reject small reflections such as those on the welds or expansion joints.

SafetyLok, 2016-12-22:
Closing due to adding rising/falling edge interrupts. But can re-open due to redundancy issues.

One idea I was playing with was timestamping using N2HET and HTU. We can get a 50bit timer at LR (Loop res) of 160ns. This would be ideal, but requires lost of work.

closed 2016-12-22 by SafetyLok