rLoopTeam/eng-software-pod #153: Allow operation with a faulty sensor, test this functionality.

omegabytes, 2016-10-29:


millipedeHz, 2016-11-03:
There'll be one redundant laser facing the i-beam...assuming I don't break one.

m4rol, 2016-11-03:
just yesterday we decided to move that one redundant i-beam distance sensor to the set of the height sensors. Now we will have four height sensors (one in each corner) and 2-ibeam sensors. Reason: After Keith tests on the brakes we no longer want to apply brakes unevenly. We will clamp them down regardless of the lateral position of the pod and let it stabilise itself. At least this is what we are going to do for the test weekend. In case we make up our mind we we still have a mounting holes prepared to move it back to face the i-beam.

closed 2016-12-14 by millipedeHz